mcu-debug/peripheral-viewer #1: Protect `main` branch

thegecko, 2023-02-06:
Can we add a branch rule to protect `main`? I'm keen to have a PR and review process in place

haneefdm, 2023-02-06:
I have never (intentionally) tried to automate a publish. I can see how to protect tags and branches, but I don't see how the publishing to the marketplace is triggered.

Yes, nothing should go in without a review...what is the process you suggest? What specifically should be enabled/disabled on the repo settings?

thegecko, 2023-02-06:
> I have never (intentionally) tried to automate a publish. I can see how to protect tags and branches, but I don't see how the publishing to the marketplace is triggered.

These are different concerns. the CI publishes the extensions.

> what is the process you suggest? What specifically should be enabled/disabled on the repo settings?

Protecting main ensures all code changes are via PRs (it blocks direct pushes to main). I would also suggest adding the CI check to ensure the CI passes before a merge can be made.