mcu-debug/peripheral-viewer #4: Announcement: Just went live in the VSCode marketplace

haneefdm, 2023-03-05:
This extension just went live. Of course, it is in *Preview* mode. A lot more testing is needed

* Works with cortex-debug SVD pack extensions
* Make sure it works in the browser and GitHub Workspaces

A release of Cortex-Debug has also been done to use this extension. It has been released as a Pre-release, so you have to opt in.

This version should be more performant on re-starts and your workspace is no longer modified to save settings

**Please file any issues as a new issue and not HERE**

@thegecko Thanks for all your help. Sorry, it took me a bit of time. Btw, I had to change the name because the marketplace would not accept `svd-viewer`. That actually delayed everything quite a bit.

Cc: @thegecko

thegecko, 2023-03-06:
> the marketplace would not accept `svd-viewer`

Probably because I already have that published, it just needs me to switch the repo on there if that name wants to be kept.

haneefdm, 2023-03-06:
Thanks, but, I will leave it as "peripheral viewer"

haneefdm, 2023-03-06:
Hi @thegecko I just published svd-viewer and its dependency. Not sure what I should test them with.

[image]

If this works out, I can publish the other extensions (rtos, memory) as well.

Someone published cortex-debug already. Two issues with this.

1. I don't own cortex-debug so it doesn't come under my mcu-debug umbrella
2. Since someone else already published it, I am guessing I can't

Also, it is out of date now and it now has dependencies that are not in open-vsx registry yet.

thegecko, 2023-03-06:
> ... I just published svd-viewer and its dependency
... it is out of date now and it now has dependencies that are not in open-vsx registry yet

My updates to this package removed the hard dependency on this tracker. In my tests it was slower and more cumbersome, so refactored it in a way that the end user could choose to use this or not. It would be used if installed and could be distributed in a cortex debug pack for this purpose if needed. Can the dependency be made soft again, please?

TBF, if you had done this in a PR I would have pointed out the drawbacks and better approach and we could have discussed it.

> Someone published cortex-debug already

Yes, I own it and the keys discussed for publishing give access to this. Again, we can collaborate on getting the right things in the right place.

What publisher and names do you want to use on the marketplaces?

haneefdm, 2023-03-06:
> My updates to this package removed the hard dependency on this tracker

Maybe you did not review the code. It is still soft just like you had it. The tracker source code is also an npm module, so we have the same logic with or without the tracker. I don't understand how it was cumbersome either and for who? I did not even mark the tracker as an extension dependency. It is a code dependency like a normal npm module. You didn't even use the npm module and declared your own interfaces.

https://github.com/mcu-debug/peripheral-viewer/blob/a14be42d5680a35bdc05762a07d769c36ff922d5/src/debug-tracker-wrapper.ts#L62

There were certain cases that were not handled right (or even crashed), it would behave differently depending on if we are using the tracker extension or direct code you had. Now, it will work the same regardless (good, bad, or ugly). If you had issues using the tracker as an npm module you should have let me know or offer to fix it. If at all possible, I want "one version of the truth"

>> Someone published cortex-debug already

> Yes, I own it and the keys discussed for publishing give access to this. Again, we can collaborate on getting the right things in the right place.
> What publisher and names do you want to use on the marketplaces?

About cortex-debug. Thanks. We cannot change the publisher name or the ID of the extension - not at least for VSCode so I don't want to modify package.json. Marus and I discussed this and we would like to not disturb the user base. Do you update the release periodically? Let me know what makes sense.

haneefdm, 2023-03-06:
I tested it both ways with or without the tracker installed. I would consider it a bug if it really needed the tracker extension for this extension to work.

thegecko, 2023-03-07:
> I did not even mark the tracker as an extension dependency.

Apologies, I thought this was an extension dependency

> You didn't even use the npm module

Glad these types are published, much cleaner 👍 

> Do you update the release periodically?

Whenever there's new changes. One approach could be to run them in parellel for a while and then deprecate `cortex-debug.svd-viewer` after a while when the peripheral viewer has stabilised?

haneefdm, 2023-03-07:
>> Glad these types are published, much cleaner 👍

Yes, I released it as an npm module since day one. Because I was using it myself and I created the option of fully instantiating the tracker of sharing an existing from an extension.

>> Do you update the release periodically?
>
> Whenever there's new changes. One approach could be to run them in parellel for a while and then deprecate cortex-debug.svd-viewer after a while when the peripheral viewer has stabilised?

Sorry, I did not ask the question properly. I meant the marus25.cortex-debug extension. The debugger that you have been releasing into Open-VSX

closed 2023-04-13 by haneefdm