mcu-debug/peripheral-viewer #21: Not compatible with VS Code 1.70.2?

joba-1, 2023-09-29:
Type: Bug Report

**Describe the bug**
- OS and Version: linux (latest tumbleweed)
- VS Code Version: 1.70.2
- Extension Version: 1.4.6
- Target Device: n/a
- Other extensions you installed (and if the issue persists after disabling them): basics for c++, cmake, jupyter, platformio, python (13)
- A clear and concise description of what the bug is.
I have not disabled all extensions, since the message clearly says what the problem is: extension is incompatible with vs code version 1.17.2.

**To Reproduce**
Steps to reproduce the behavior: just try to install the extension...
[image]

**Expected behavior**
extension installs

joba-1, 2023-09-29:
Context: I'd like to use the Cortex-Debug extension to debug a raspberry pico, but installation depends on this one :(

joba-1, 2023-09-29:
fwiw: I downloaded the 1.4.6 release vsix, unzipped it, changed the required version in the manifest zipped the vsix again and installed it. So far, debugging seems to work...  :)

MatthijsReyers, 2023-09-30:
I have exactly this issue as well, I simply upgraded my vscode installation to the lastest version, but if there is no reason for the extension to force versions `"^1.75.0"` it might be worth downgrading it?

joba-1, 2023-09-30:
all available releases have the same minimum requirement.

But I must admit, the real problem for me was that the repository refresh was turned off for the vs code repo.
Now it is on and I am at code v1.82.2. All good :)

closed 2023-09-30 by joba-1