mcu-debug/peripheral-viewer #23: Register array name placeholder replaced with "undefined" instead index

kossnikita, 2023-10-19:
Type: Bug Report

**Describe the bug**
- OS and Version: W11 Pro
- VS Code Version: 1.76.2
- Extension Version: 1.4.6

`%s` register name placeholder replaced with `undefined` instead index.

**To Reproduce**
Steps to reproduce the behavior:
1. Create array of registers with `%s` in name
2. Check it's name in extension

**Expected behavior**
Names `OM0`, `OM1`... `OM15` for this case

**Code sample and logs**

```xml
<register>
  <name>OMODE</name>
  <displayName>OMODE</displayName>
  <description>GPIO output mode register</description>
  <addressOffset>0x4</addressOffset>
  <size>0x20</size>
  <access>read-write</access>
  <resetValue>0x00000000</resetValue>
  <fields>
    <field>
      <dim>16</dim>
      <dimIncrement>0x1</dimIncrement>
      <dimIndex>0-15</dimIndex>
      <name>OM%s</name>
      <description>GPIOx pin %s outpu mode configurate</description>
      <bitOffset>0</bitOffset>
      <bitWidth>1</bitWidth>
      <enumeratedValues>
        <name>OM0</name>
        <usage>read-write</usage>
        <enumeratedValue>
          <name>PushPull</name>
          <description>Push-pull</description>
          <value>0</value>
        </enumeratedValue>
        <enumeratedValue>
          <name>OpenDrain</name>
          <description>Open-drain</description>
          <value>1</value>
        </enumeratedValue>
      </enumeratedValues>
    </field>
  </fields>
</register>
```

**Screenshots**
[image]

**Additional context**
[Full SVD](https://github.com/kossnikita/at32f4xx-pac/blob/master/src/at32f421/at32f421.svd)

kossnikita, 2023-11-08:
Perhaps the problem is

```xml
<dimIndex>0-15</dimIndex>
```

Because non-zero indexes don't cause such a problem.

EGOKM, 2024-01-03:
Hello,
well, I observed the same Problem ;-), changing the Start-Value to non-0 removes the "undefined" BUT this is in fact not the point of the matter ;-) , so it would be very nice and recommended to fix this Bug in the near future ...
Normally I think this is not too complicated todo - if I can help to fix this Bug I will try, but I need some little instructions where to find all those svd viewer code ;-),
THX a lot, Enrico

EGOKM, 2024-01-04:
Hello again,
I've found some issue while checking dimIndex - Range, made a PR ... once approved I hope it will work ;-)

kossnikita, 2024-03-01:
@thegecko could you please help us with this? This issue is so annoying

kossnikita, 2024-03-05:
@haneefdm sorry for the ping. Are there any active developers here?

haneefdm, 2024-03-05:
Yes, but I have been relocating for the last two months. Not yet settled. I will look at this in a week or two

kossnikita, 2025-08-30:
Fixed by https://github.com/mcu-debug/peripheral-viewer/pull/27